nkitanov/iaq_board #3: JLC PCB does not support Gerber X2

fernplant, 2021-01-07:
Hi. Fantastic project, thanks for sharing. 
I went to JLCPCB to order some PCBs and this was the email they sent to me after my order:

> We noticed your files are in Gerber X2 format but currently JLCPCB is still not ready for Gerber X2 (We are working on to support Gerber X2, but not fully done yet).
>
> If we use current CAM tools to convert the X2 file, there is high risk of lossing traces and get deformed pads.
> 
> Some customers already received defective boards with Gerber X2 files.
> 
> So please do not check "Use extended X2 format" when you plot the Gerber files.
> 
> Ref:
> 
> https://forum.kicad.info/t/jlcpcb-strange-issue-with-rounded-rectangle-pads-kicad-nightly/26715
> 
> When we are ready for Gerber X2, we'll announce it.

adebree, 2021-01-07:
That's weird, I just received a batch of 10 PCB of the IAQ Board from
them... I used their normal online ordering tool.

This message was sent from a mobile device

nkitanov, 2021-01-07:
@fernplant did you used the files I provided or you regenerated them? I didn't had issues with that in the past but it was in March when I ordered the last time. I can update the files...

fernplant, 2021-01-13:
Files uploaded to them were as provided in plot.zip . Maybe was just a fluke?

Badn1nja, 2021-01-31:
I'm going to try to use the instructions here to generate the files again.

I can see they want one set of files where there is two in the files.
https://support.jlcpcb.com/article/44-how-to-export-kicad-pcb-to-gerber-files

closed 2022-01-26 by nkitanov