nkitanov/iaq_board #4: Firmware not compatible with Home Assistant ESPHome addon

Dinth, 2021-01-16:
Hi. Could you please provide a config file which would be compatible with ESPHome docker image (HA addon)?
Right now it requires: brightness.yaml include to be pasted to the main config and also is missing fonts, which im not sure how to overcome.

mikenabhan, 2021-01-17:
Brightness.yaml is linked in the firmware folder.

https://github.com/nkitanov/iaq_board/blob/master/firmware/includes/brightness.yaml

The fonts folder is also included.

https://github.com/nkitanov/iaq_board/tree/master/firmware/fonts

If you download the firmware folder so the paths match, the firmware will compile.

adebree, 2021-01-21:
For `iaq_board.yaml` to compile in the ESPHome HA Addon I had to also add a couple dependencies to the yaml:

```
  platformio_options:
    lib_deps: 
      - NeoPixelBus@2.6.0
      - ArduinoJson-esphomelib@5.13.3
      - AsyncTCP-esphome@1.1.1
      - ESPAsyncWebServer-esphome@1.2.7
 ```

(The NeoPixelBus was already defined as a dep)

Dinth, 2021-01-21:
> For `iaq_board.yaml` to compile in the ESPHome HA Addon I had to also add a couple dependencies to the yaml:
> 
Thanks. Im still not sure, how i can deal with fonts, HA addon doesnt allow to upload any external dependencies like fonts

adebree, 2021-01-21:
You need to manually upload the font and the brightness.yaml into the config directory of the addon. It depends a lot on how you actually are running home-assistant. But you can either SSH into HA or use a addon like Samba to get filesystem access  to it. For me that config directory is actually called `esphome` and lives within my Home Assistant data dir. The config files contains the yaml files you have created via the web interface of the ESPHome addon. 

In that same dir (so next to where you have `iaq_board.yaml` and all the other ESPHome device configs) you create a `fonts` dir and upload the `Arials.ttf` file. And you create an `includes` dir where you put the `brightness.yaml` file in.

After adding the earlier mentioned additional dependencies you should be able to compile and manage the device.

closed 2021-01-21 by Dinth