nkitanov/iaq_board #5: Question about turning the display off an the second button to reset the baseline of the voc sensor

sebastianf, 2021-01-17:
Hi,
thank you for providing your code. It is is great!

I have a similar setup and I am new to ESPHome, so I wanted to peek in your code.
I can't see where you turn off the display and where you configure the second button to reset the baseline for the voc sensor.

Can you point me in the right direction?

Thanks!

nkitanov, 2021-01-18:
Hi,

Here is the code to turn off the LEDs, it's raw C++ included with lambda. I couldn't find any other way although I am sure there is more elegant way to do it. 

For the calibration of CO2 sensor (I think you are speaking for it and not VOC sensor, as VOC sensor baseline cannot be externally calibrated, it's hardcoded) here is the code, it's just a switch which executes calibration:
```
switch:
  - platform: template
    name: "CO2 Sensor Zero Calibration"
    turn_on_action:
      - mhz19.calibrate_zero: mh_sensor
      - switch.turn_off: co2_cal
    id: co2_cal
```

sebastianf, 2021-01-18:
Thanks a lot.
I found this for turning an OLED Display off
```
lambda: |-
          it.fill(COLOR_OFF);
```

nkitanov, 2021-01-18:
Ah, nice, this can be used also in the code instead of decreasing the brightness as I do it in a dumb way.

closed 2022-01-26 by nkitanov