nkitanov/iaq_board #6: Troubleshooting, all of the I2C devices give 'communication failed'

adebree, 2021-01-20:
Thanks for your awesome project, soldering it was really fun. Unfortunately I am now facing the issue none of the I2C devices can be found. I did a fully assembled board, so with the TSL2561, the BME280, both the SSD1306 OLEDs and the SGP30. In the ESPHome Debug log I get:

```
[E][bme280.sensor:146]: Communication with BME280 failed!
[E][tsl2561:044]: Communication with TSL2561 failed!
[E][ssd1306_i2c:031]: Communication with SSD1306 failed!
[E][ssd1306_i2c:031]: Communication with SSD1306 failed!
```

All the soldering looks good, have you any suggestion on how to approach this problem?

nkitanov, 2021-01-21:
Can you check if you properly soldered the receptacle or pins for the ESP board? Also another potential issue if you short circuit i2c pins bellow the TLS sensor (I done it once). It can make the whole i2c bus unusable which is your case, since none of the sensor is found.. Soldering light sensor is a bit of tricky thing and can lead to issues. I would check with a multimeter if there is any short circuit to gnd or power of the i2c pins. Or short circuit between SCL and SDA. If that is the case, just properly resolder the TLS sensor which can be a fun with regular solder iron. :)

adebree, 2021-01-21:
Your hunch was correct! It was the TLS sensor, after desoldering the I2C devices started working. Unfortunately I destroyed the TLS during desoldering. 

The reason I desoldered is that I started to doubt the orientation of the TLS. I studied the pictures to find a clue how it should be oriented. Can you explain how the TLS should be oriented so when I get a replacement I won't make the same mistake again? Thanks!

nkitanov, 2021-01-21:
You can check in the datasheet how to orient it: https://github.com/nkitanov/iaq_board/blob/master/kicad/iaq_device/datasheets/TSL2561.pdf

Also using a magnifying glass is a good idea.

adebree, 2021-01-21:
ah! Got it, pin1 has a square / rectangle copper pad while the other 5 have round. Excellent, thanks!

closed 2021-01-21 by adebree