nkitanov/iaq_board #7: fatal error: ArduinoJson.h: No such file or directory

jamarju, 2021-01-31:
Here's the result of `esphome iaq_board.yaml run`.

Tried with:

* ESPHome v1.15.3
* ESPHome v1.17.0-dev

```
$ esphome iaq_board.yaml run
INFO Reading configuration iaq_board.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d iaq_board
Processing iaq_board (board: wemos_d1_mini32; framework: arduino; platform: espressif32@1.12.4)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Dependency Graph
|-- <NeoPixelBus> 2.6.0
|   |-- <SPI> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Wire> 1.0.1
|-- <Update> 1.0
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
Compiling .pioenvs/iaq_board/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/iaq_board/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/iaq_board/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/iaq_board/src/esphome/components/api/proto.cpp.o
In file included from src/esphome/components/light/light_color_values.h:8:0,
                 from src/esphome/components/light/light_effect.h:4,
                 from src/esphome/components/light/light_state.h:6,
                 from src/esphome/core/controller.h:11,
                 from src/esphome/components/api/util.h:5,
                 from src/esphome/components/api/list_entities.h:5,
                 from src/esphome/components/api/list_entities.cpp:1:
src/esphome/components/json/json_util.h:4:25: fatal error: ArduinoJson.h: No such file or directory

*********************************************************************
* Looking for ArduinoJson.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ArduinoJson.h"
* Web  > https://platformio.org/lib/search?query=header:ArduinoJson.h
*
*********************************************************************

compilation terminated.
In file included from src/esphome/components/light/light_color_values.h:8:0,
                 from src/esphome/components/light/light_effect.h:4,
                 from src/esphome/components/light/light_state.h:6,
                 from src/esphome/core/controller.h:11,
                 from src/esphome/components/api/util.h:5,
                 from src/esphome/components/api/proto.cpp:2:
src/esphome/components/json/json_util.h:4:25: fatal error: ArduinoJson.h: No such file or directory

*********************************************************************
* Looking for ArduinoJson.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ArduinoJson.h"
* Web  > https://platformio.org/lib/search?query=header:ArduinoJson.h
*
*********************************************************************

compilation terminated.
In file included from src/esphome/components/light/light_color_values.h:8:0,
                 from src/esphome/components/light/light_effect.h:4,
                 from src/esphome/components/light/light_state.h:6,
                 from src/esphome/core/controller.h:11,
                 from src/esphome/components/api/api_server.h:4,
                 from src/esphome/components/api/api_server.cpp:1:
src/esphome/components/json/json_util.h:4:25: fatal error: ArduinoJson.h: No such file or directory

*********************************************************************
* Looking for ArduinoJson.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ArduinoJson.h"
* Web  > https://platformio.org/lib/search?query=header:ArduinoJson.h
*
*********************************************************************

compilation terminated.
*** [.pioenvs/iaq_board/src/esphome/components/api/api_server.cpp.o] Error 1
*** [.pioenvs/iaq_board/src/esphome/components/api/list_entities.cpp.o] Error 1
*** [.pioenvs/iaq_board/src/esphome/components/api/proto.cpp.o] Error 1
In file included from src/esphome/components/light/light_color_values.h:8:0,
                 from src/esphome/components/light/light_effect.h:4,
                 from src/esphome/components/light/light_state.h:6,
                 from src/esphome/core/application.h:30,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/json/json_util.h:4:25: fatal error: ArduinoJson.h: No such file or directory

*********************************************************************
* Looking for ArduinoJson.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ArduinoJson.h"
* Web  > https://platformio.org/lib/search?query=header:ArduinoJson.h
*
*********************************************************************

compilation terminated.
*** [.pioenvs/iaq_board/src/esphome/components/api/api_connection.cpp.o] Error 1
============================================================================== [FAILED] Took 2.86 seconds ==============================================================================
```

Badn1nja, 2021-01-31:
Hi thats a complication error. Please read the last of that error where platformio is offering to install the library

Badn1nja, 2021-01-31:
https://platformio.org/lib/show/64/ArduinoJson

jamarju, 2021-01-31:
Thanks for the fast response! But how should I go about installing that library?

I ran this from the `firmware` dir:

```
$ pio lib install "bblanchon/ArduinoJson"
*****************************************************************************************************************************************************************************************
Obsolete PIO Core v4.3.4 is used (previous was 5.0.4)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
*****************************************************************************************************************************************************************************************
Library Storage: /Users/javi/V/DEV/esphome/iaq_board/iaq_board/.pio/libdeps/iaq_board
LibraryManager: Installing ArduinoJson
git version 2.24.3 (Apple Git-128)
Cloning into '/Users/javi/V/DEV/esphome/iaq_board/iaq_board/.pio/libdeps/iaq_board/_tmp_installing-an_ept8o-package'...
remote: Enumerating objects: 470, done.
remote: Counting objects: 100% (470/470), done.
remote: Compressing objects: 100% (413/413), done.
remote: Total 470 (delta 62), reused 160 (delta 11), pack-reused 0
Receiving objects: 100% (470/470), 309.86 KiB | 1.17 MiB/s, done.
Resolving deltas: 100% (62/62), done.
ArduinoJson @ ab90212 has been successfully installed!
```

But I'm still getting the same error:

```
$ esphome iaq_board.yaml run
INFO Reading configuration iaq_board.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d iaq_board
*****************************************************************************************************************************************************************************************
Obsolete PIO Core v4.3.4 is used (previous was 5.0.4)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
*****************************************************************************************************************************************************************************************
Processing iaq_board (board: wemos_d1_mini32; framework: arduino; platform: espressif32@1.12.4)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Dependency Graph
|-- <NeoPixelBus> 2.6.0
|   |-- <SPI> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Wire> 1.0.1
|-- <Update> 1.0
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
Compiling .pioenvs/iaq_board/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/iaq_board/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/iaq_board/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/iaq_board/src/esphome/components/api/proto.cpp.o
In file included from src/esphome/components/light/light_color_values.h:8:0,
                 from src/esphome/components/light/light_effect.h:4,
                 from src/esphome/components/light/light_state.h:6,
                 from src/esphome/core/application.h:30,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/json/json_util.h:4:25: fatal error: ArduinoJson.h: No such file or directory

*********************************************************************
* Looking for ArduinoJson.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ArduinoJson.h"
[... 40 lines cut ...]

compilation terminated.
*** [.pioenvs/iaq_board/src/esphome/components/api/list_entities.cpp.o] Error 1
In file included from src/esphome/components/light/light_color_values.h:8:0,
                 from src/esphome/components/light/light_effect.h:4,
                 from src/esphome/components/light/light_state.h:6,
                 from src/esphome/core/controller.h:11,
                 from src/esphome/components/api/util.h:5,
                 from src/esphome/components/api/proto.cpp:2:
src/esphome/components/json/json_util.h:4:25: fatal error: ArduinoJson.h: No such file or directory

*********************************************************************
* Looking for ArduinoJson.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ArduinoJson.h"
* Web  > https://platformio.org/lib/search?query=header:ArduinoJson.h
*
*********************************************************************

compilation terminated.
*** [.pioenvs/iaq_board/src/esphome/components/api/proto.cpp.o] Error 1
============================================================================== [FAILED] Took 2.37 seconds ==============================================================================
```

(I'm on esphome v1.15.3 now).

Badn1nja, 2021-02-01:
Hi, It looks like ESPHome isn't detecting the dependency. this is an issue with ESPHome and not IAQ_Board can you have a look at this and see if it helps? https://github.com/esphome/issues/issues/993. or https://github.com/esphome/esphome-core/issues/112

I'm not flashing this yet so I can't replicate here but it's likely a Platformio issue as stated in the links above... IIRC Platformio uses arduino ide libarys so maybe they need to be installed there first?

Badn1nja, 2021-02-01:
Having a second look...

Follow this guide; https://github.com/hajdbo/ESPHome-Core

You need these libaries by the look of it

    https://github.com/marvinroger/async-mqtt-client/archive/master.zip
    https://github.com/OttoWinter/ArduinoJson/releases
    https://github.com/FastLED/FastLED/releases
    https://github.com/me-no-dev/ESPAsyncWebServer/archive/master.zip
    https://github.com/me-no-dev/AsyncTCP/archive/master.zip (only needed for ESP32)
    https://github.com/me-no-dev/ESPAsyncTCP/archive/master.zip (only needed for ESP8266)

After installing esphomelib, you will find a variety of example sketches under File > Examples > esp

jamarju, 2021-02-03:
Thanks for the hints. I think I made progress.

I added this under `platformio_options`:

```
  platformio_options:
    lib_deps:
      - NeoPixelBus@2.6.0
      - ArduinoJson-esphomelib@5.13.3
      - https://github.com/OttoWinter/AsyncTCP
      - https://github.com/OttoWinter/ESPAsyncWebServer
```

I think it now resolves all the dependencies correctly but I'm still getting a strange error at the end. It may be because I don't have the ESP32 physically connected so I'll try that later and report back.

```
$ esphome iaq_board.yaml run
INFO Reading configuration iaq_board.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d iaq_board
*****************************************************************************************************************************
Obsolete PIO Core v4.3.4 is used (previous was 5.0.4)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
*****************************************************************************************************************************
Processing iaq_board (board: wemos_d1_mini32; framework: arduino; platform: espressif32@1.12.4)
-----------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Dependency Graph
|-- <NeoPixelBus> 2.6.0
|   |-- <SPI> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <AsyncTCP-esphome> 1.1.1 #c18b7ca
|-- <ESPAsyncWebServer-esphome> 1.2.7 #32ab7bd
|   |-- <AsyncTCP-esphome> 1.1.1 #c18b7ca
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Wire> 1.0.1
|-- <Update> 1.0
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
Building .pioenvs/iaq_board/firmware.bin
Retrieving maximum program size .pioenvs/iaq_board/firmware.elf
Checking size .pioenvs/iaq_board/firmware.elf
RAM:   [=         ]  13.6% (used 44584 bytes from 327680 bytes)
Flash: [======    ]  58.4% (used 1071890 bytes from 1835008 bytes)
Traceback (most recent call last):
  File "/Users/javi/.platformio/packages/tool-esptoolpy@1.20600.0/esptool.py", line 57, in <module>
    import serial.tools.list_ports as list_ports
  File "/Users/javi/opt/miniconda3/envs/esphome/lib/python3.7/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Users/javi/opt/miniconda3/envs/esphome/lib/python3.7/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Users/javi/opt/miniconda3/envs/esphome/lib/python3.7/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
*** [.pioenvs/iaq_board/firmware.bin] Error 1
================================================ [FAILED] Took 5.05 seconds ================================================
```

nkitanov, 2021-02-04:
Keep in mind that I build this config with probably older version of esphome, I will add as ac omment later the version I use to build. However if you use `platformio_options` you can get errors on the first build. I remember that I had this and found out it's normal. Also I added this library dependance, because bug in neopixel 2.5.7 library, which is fixed in 2.6.0. Have to check if this library is updated in the latest esphome and if so, then `platformio_options` can be removed.

nkitanov, 2021-02-06:
I tested now with latest esphome 1.16.0 and it still uses neopixel 2.5.7 so lib deps cannot be removed. I updated yaml file to include comment with esphome version compatible with it.

nkitanov, 2021-02-06:
This is the issue in esphome for this: https://github.com/esphome/issues/issues/1481

jamarju, 2021-02-07:
Thanks @nkitanov !! It works now.

Now when I plug the esp32 I can't see anything on the display and the LEDs don't turn on either, but I guess that's an unrelated issue so you can close this one.

nkitanov, 2021-02-07:
Yes, this is how it works. I saw in the esphome issue someone added merge request to add neopixel 2.6.1, so it will be fixed soon,.

closed 2022-01-26 by nkitanov