nkitanov/iaq_board #9: can a MH-Z19C be used as a drop in for the MH-Z19B?

SyedAmerGilani, 2021-02-08:
The supplier in the BOM for the MH-Z19B is out of stock. It is much easier to get an MH-Z19C here in Germany from a trustier supplier then a MH-Z19B. 
From the Datesheet the MH-Z19C seems to be pin compatible, but i have seen reports that say that it is more fickle with its voltage supply.
Do you have any Idea if it should work on this PCB?

nkitanov, 2021-02-08:
As I checked quickly the specs - it should be fine if the pinout is the same. Voltage requirement is 5V+/-0.1V which is fine. This sensor is powered directly by 5V from the USB conneced on the ESP board. USB voltage is in this range. Can you show some source where voltage issues are commented.

SyedAmerGilani, 2021-02-08:
See here in the lower part of the Post: https://emariete.com/en/power-from-nodemcu-wemos-mini-d1-and-arduino/
The Concern seems to be that the USB voltage can dip below 4.9V

nkitanov, 2021-02-08:
Yes, it's looking the power supply requirement is more strict than on the older versions. Just measured my device with USB from a PC and it shows 4.75V - there is no regulation of 5V rail so it have what it is provided from the USB. Theoretically this one cannot work with the new MH-Z19C, but practically it can and it have to be tested. However you will be running the senors with out of spec power supply. I would stick to the MH-Z19B for now, since this model is still available. Making 5V from 4.7V needs extra circuitry and does not worth the effort for the moment.

SyedAmerGilani, 2021-02-08:
ok i agree. Thank you !
so now to look which aliexpress dealer to trust that it does not sell fake MH-Z19B which seem to be in circulation...

closed 2021-02-08 by SyedAmerGilani