nkitanov/iaq_board #10: SGP30 - Unkown setup error

darkson95, 2021-02-25:
Hi,

First of all, I would like to thank you for the cool project. Super work! 😎

Unfortunately I have a problem with the SGP30 sensor. I built and programmed everything as you described in the repository, but when starting the device I get the following message:
```yaml
[22:24:58][C][sgp30:178]: SGP30:
[22:24:58][C][sgp30:179]:   Address: 0x58
[22:24:58][W][sgp30:195]: Unknown setup error!
[22:24:58][C][sgp30:209]:   Update Interval: 1.0s
[22:24:58][C][sgp30:210]:   eCO2 'eCO2'
[22:24:58][C][sgp30:210]:     Unit of Measurement: 'ppm'
[22:24:58][C][sgp30:210]:     Accuracy Decimals: 1
[22:24:58][C][sgp30:210]:     Icon: 'mdi:molecule-co2'
[22:24:58][C][sgp30:211]:   TVOC 'TVOC'
[22:24:58][C][sgp30:211]:     Unit of Measurement: 'ppb'
[22:24:58][C][sgp30:211]:     Accuracy Decimals: 1
[22:24:58][C][sgp30:211]:     Icon: 'mdi:radiator'
[22:24:58][C][sgp30:213]:   Compensation:
[22:24:58][C][sgp30:214]:     Temperature Source: 'Temperature'
[22:24:58][C][sgp30:214]:       Unit of Measurement: '°C'
[22:24:58][C][sgp30:214]:       Accuracy Decimals: 1
[22:24:58][C][sgp30:214]:       Icon: 'mdi:thermometer'
[22:24:58][C][sgp30:215]:     Humidity Source: 'Humidity'
[22:24:58][C][sgp30:215]:       Unit of Measurement: '%'
[22:24:58][C][sgp30:215]:       Accuracy Decimals: 1
[22:24:58][C][sgp30:215]:       Icon: 'mdi:water-percent'
```

The first thing I thought was that the sensor is broken and I tried a NodeMCU (ESP8266) and set it up in ESPHome (+ Home Assistant). _There everything worked and I got legitimate values back, so the sensor can not be broken._

This is the relevant configuration for the SGP30:
```yaml
i2c:
  scl: GPIO22
  sda: GPIO21
  frequency: 100kHz

sensor:
  - platform: sgp30
    address: 0x58
    eco2:
      name: "eCO2"
      accuracy_decimals: 1
      id: eco2
    tvoc:
      name: "TVOC"
      accuracy_decimals: 1
      id: tvoc
    update_interval: 1s
    baseline:
      eco2_baseline: 0x92B7
      tvoc_baseline: 0x935A
    compensation:
      temperature_source: temp
      humidity_source: hum
```

Do you guys have any idea what the problem could be?

nkitanov, 2021-02-25:
Hi,

I have similar error with my second SGP30 from Aliexpress and decided that is broken so I  throw it in the bin. Then yesterday it was reported to me for a simliar issue which can be fixed by removing some of the orther i2c devices like bme280 sensor - then magically sgp30 sensor is working. The setup there is all things on i2c - 2 displays, bme280 and sgp30. 

I have a hypothesys that the 4 paralel pullup resistors are dropping too much the voltage state on the bus. Schematic of the sgp30 brearout is similar to this: https://cdn-learn.adafruit.com/assets/assets/000/097/500/original/adafruit_products_SGP30_sch.png?1607379580

If you are using all 4 devices on the i2c, can you try removing any of it to see if sgp30 is working. Unfortunatelly I don't have anymore my problematic sgp30 to measure the bus with the scope and see it that can cause it. If it's working then - easy try is to remove some of the pullups on any of the other i2c devices. They normally are 10 kOhm and 4 in paralel should not cause issues but who know...

darkson95, 2021-02-25:
Thank you for the quick response at this late hour.

I have removed all displays and also commented out in the code. The sensor then returned me values. So I first put the big display back on to check if it still works. Then the small display.

For whatever reason all sensors and all displays now work as I wanted them to. Thanks for the support! 🥇

nkitanov, 2021-02-25:
OK. I will reopen that, as I thing have to check it when I get back some new sgp30 sensors.  :)

darkson95, 2021-03-10:
Hi again! :)

Today the ambientlight sensor TSL 2561 arrived and I have now installed all the parts as you described in the project. Unfortunately, I have the problem that the SGP30 can not start again, so I have removed all displays that at least data can be captured.

I thought I'd share this with you if you still have the issue open. 🙂

nkitanov, 2021-03-10:
Hi. I do not havy any sgp30 sensors right now, and when I reveive a few in 2 weeks will do verification. Probably removing the pullup resistors on the sgp30 board will solve it.

nkitanov, 2021-03-14:
Ok, I got a few SGP30 breakout boards from Aliexpress. As expected the first one I tested gives the same issues. Then I hooked up the osciloscope to check the i2c levels but it's not a problem, this board have 10k pullups so no issues there. SGP30 is using 1,8V power supply and on this breakout it is having the chinese LDO regulator [SC6206B-65K5](https://lcsc.com/product-detail/Dropout-Regulators-LDO_Shenzhen-Fuman-Elec-SC6206B-65K5_C88294.html). Anyway I noticed that the regulated voltage after the LDO is having very unusual 440mVp-p noise, which on my working SGP30 board is not there. Here is a snap from the osciloscope:

[image]

Then I measured the output capacitor after the LDO regulator and it was aroung 2uF, while on the working board it was 10 uF, although even 2uF should work. Then I removed the output capacitor and the board started working. 

My not very detailed conclusion, since I am not electronics expert: These chinese breakout board are pretty bad in quality. Apparently they soldered what they have. Also I guess some of these capacitors are faulty. I got a few other boards and some of them have the same issue, although even with the noise some were working but now all. There was one completely dead probably because of faulty mosfet array (the switch circuit to drop i2c signals from 3.3V to 1.8V). I have a friend who found on his SGP30 board short circuit in the LDO regulator pins. So it's a gambling if this thing will work.

What is the solution if your SGP30 sensor experience issues (either completely blocking i2c like not having any displays, etc. or just not receiving TVOC data)? Just unsolder this capacitor and if you have 10 uF SMD capacitor put it there or just leave it without (still working and noise not there anymore):

[image]

If that is not solving the issue, then there is somethinge else with these poor boards. You can measure the LDO if it's providing 1.8V, check the i2c after the mosfet array (having proper 1.8V high levels), etc... Check for short circuits, etc. I might change the design to use more simplistic SGP30 boards (without the requlator) but then the 1.8V regulation and drop on i2c signals needs to be utilized on my board. And probably not a good idea - it increases the complexity and TVOC is not so important anyway (at least for me).

kokanov-del, 2021-03-29:
Hi!
I also thank you for the compact project!
The SGP30 module was also a problem for me.
After removing the BME280, SGP30 - did it work !?
The treatment offered by you restored the functionality of the SGP30 module !!
I later found the following.
Without a 10μf capacitor, the SGP30 sensor module is not sensitive to VOS and eCO2.
With 10μf capacitor, works normally.
Thanks for the support!

[image]
[image]
[image]

donnie-darko, 2021-04-13:
Just a quick head-up:
maybe this one is a better SGP30 Breakout board with quality components:
https://shop.pimoroni.com/products/sgp30-air-quality-sensor-breakout

Anyway, thanks for this cool project! I find nowhere else an easy, low-cost DIY air quality monitor like this one

nkitanov, 2021-04-13:
This is better quality for sure. The only problem here is that this board is not fitting the layout on the iaq board. You need to connect it in a weird way or with floating cables.

donnie-darko, 2021-04-14:
Or maybe consider this breakout board when designing the next version of the PCB layout 😸

beicnet, 2021-09-30:
@nkitanov I like your project and I wanted to use the same SGP30 from Ali, so, today I got two of them, but unfortunately neither of them are working with other I2C devices attached.

The SGP30 (GY-SGP) Sensor is working without any issue if there is no other I2C devices attached to the bus.

If I remove the capacitor what you suggested, the sensor module will not work as expected (will return wrong Device ID and Values too).

Did you got to some other conclusion over the time past by?

nkitanov, 2021-09-30:
@beicnet I don't know. The only thing I can suggest - try soldering 0.1uF bypass capacitor on the place of the removed one and try again.

beicnet, 2021-09-30:
@nkitanov I made a quick test on UNO with this SGP30 sensor board:

It can work only with two other i2c devices beside, if you add a third device it will stop and block the whole i2c bus.

What do you think?

beicnet, 2021-09-30:
@nkitanov It was all pull-up resistor caused issue (fixed).

mwolter805, 2022-01-01:
Build IAQ-Board with SGP30 ordered December 8th 2021 from this [link](https://www.aliexpress.com/item/4000004614708.html) and did not have any issues.

nkitanov, 2022-01-07:
> Built an IAQ-Board with an SGP30 ordered December 8th, 2021 from this [link](https://www.aliexpress.com/item/4000004614708.html) and did not have any issues with an BME280 connected.

Sometimes it works, I guess. :)

closed 2022-01-07 by nkitanov

divB0, 2023-02-25:
I am having a similar issue. SGP30 with BME280 doesn't work. @beicnet how did you fix the issue?

olmelabs, 2024-03-01:
I had the same issue when powering SGP30 from AliExpress from 5v. As an experiment, I put it to 3.3v pin and magically it started working. Documentation states it is either 3.3 or 5v, but seems not for this guy.

mwolter805, 2024-03-02:
Could be a power supply issue. I'm using a different dev board, [QUINLED-ESP32 ABE](https://quinled.info/quinled-esp32/) where the 5v is powered from PoE and has stronger 3v3 than typical.