nkitanov/iaq_board #11: Second Display linked in BOM does not work correctly

SyedAmerGilani, 2021-03-22:
I bought the exact linked display. But i took the v2 Version with the Buttons on the Ali Express site.
The Problem is that it displays the same contents as the first display, regardless which address is configured.

Which looks like this:
[image]

The Display looks like this:
[image]

I even tried removing the  first display completely from the code which just had the effect that none of the displays showed anything.
My current assumption is that those displays are simply bad, but maybe somebody has an input.

Funny thing about the first display is that mine is configurable on the back between 0x78 and 0x7A but it still works with the default addresses.

nkitanov, 2021-03-22:
It is simple - both displays have the same i2c address. This particular display have a tiny connection between the two pads (I got that info from a friend, never had these displays). Just cut it with a razor and it should work:  

[image]

SyedAmerGilani, 2021-03-23:
oh :man_facepalming: 
what i thought was what they meant is that those pads should be soldered or unsoldered...
just measured them, yeah they are connected even then unsoldered.
thanks for the input.

closed 2021-03-23 by SyedAmerGilani

nkitanov, 2021-03-23:
I think I also have one of these somewhere and maybe year ago had the same issue. :) These Chinese parts are like a box of chocolates. You never know what you're gonna get. :)