nkitanov/iaq_board #12: Questions about this device

AalianKhan, 2021-04-14:
Hello,
I am building a device for room presence detection and I was wondering if this type of device might be what I need for the job. 
From asking around I found out that humans produce lots of carbon dioxide when breathing so it can become a viable metric for presence detection. 
So can you or someone who built this device can tell me the carbon levels when there is a human present and when there is no one. I am mainly interested in how big the difference is. 
Also how much time does it take for the carbon levels to be normal/stable. 

And about the carbon dioxide sensors you said that the sgp30 is more or less inaccurate compared to the dedicated sensor so I was wondering how inaccurate is it?
Thank you!

nkitanov, 2021-04-14:
Hi. Why you want to detect presense with CO2 senor? You can use more traditional approach with PIR sensors. You can detect eventually presense with CO2 senor but it's very odd - it will be slow to react, it depends on the number of people and size of the room, also internal ventilation, etc. It's much better to use PIR or bluetooth tokens and similar more traditional approach.

AalianKhan, 2021-04-14:
I will use PIR sensors for detecting the the motion but when the person is not moving, detecting CO2 may be a good way to detect if someone is present. 

Also I am looking for a way to detect people without devices that people have to carry for example when guests come over and such.

nkitanov, 2021-04-15:
You can but it's really a side effect and dependant of a lot of things. For example in a not ventilated room the CO2 will increase when somebody is there but if you ventilate it - it drops. I would rather use IR long range sensor to detect that there is unmoving person. It's more complicated but much more stable.

AalianKhan, 2021-04-15:
I see, thank you for your advice. I think I will start looking into esp32
cams and object detection.

closed 2022-01-26 by nkitanov