nkitanov/iaq_board #13: PMSX003 Sleep mode

jcastro, 2021-05-06:
Hello!
Amazing work on the sensor. I've got already all the components currently working on a protoboard.

I've got a question related the PMSX003 (I have the PMS7003). I noticed that when sleeping the fan is audible for me so I was wondering if there's a way to put it into sleep mode and wake it up each XX minutes to take a reading. That would also hopefully make the sensor life longer.

I've checked the yaml in the repo but it seems there's nothing like that included.

Thanks for any info and sharing this with us!

nkitanov, 2021-05-06:
Hi,

Actually it's there, but commented: https://github.com/nkitanov/iaq_board/blob/b1ebee55ac306f0fe6a9aa39c6b358623c784f9c/firmware/iaq_board.yaml#L619

There is a switch pms_set which is always on, but you can uncomment this setting and tweak it. If you uncomment it it will run for 10s each 60s. However keep in mind that the fan needs to work sometime in order to get correct data. I think 10s is enough. But I don't use it on any of my devices. You can tweak it and rebuild the binary if you want to turn it on and off. Also by having this pms_set switch you can turn it on demand by the home assistant (like only when you need to know the PM values).

P.S. you have very good hearing, or the PMS sensor is faulty. :) I had a device 40cm from my head and I do not hear it during the night.

jcastro, 2021-06-16:
Ohhh nice! Will give it a try. Thanks so much!

closed 2021-06-16 by jcastro

jcastro, 2021-06-16:
Alright, I got it working. I don't have the full device just a few sensors connected through dupont cables on the breakout board. Looking at the schematics there's a pms_set that also needs to be connected but I guess in that case I will have to use the other connectors and not the dupont! :(

[image]