nkitanov/iaq_board #20: No reading from PMS7003

DrBlokmeister, 2022-02-27:
Hey! First of all thanks for your work on this project. I really appreciate it. It's the first time I ordered a custom PCB and I hope it will bring me to making more custom devices.

I have ordered your PCB, with the D1 mini ESP32, the MH-Z19 and PMS7003. Other sensors/components are still on their way. 

Therefore I use a limited part of your config:
'''YAML
uart:
  - rx_pin: GPIO1
    baud_rate: 9600
    id: pms
  
  - rx_pin: GPIO18
    tx_pin: GPIO19
    baud_rate: 9600
    id: mh

sensor:
    - platform: pmsx003
      type: PMSX003
      uart_id: pms
      pm_1_0:
        name: "PM 1 Concentration"
        id: pm1
      pm_2_5:
        name: "PM 2.5 Concentration"
        id: pm2_5
      pm_10_0:
        name: "PM 10 Concentration"
        id: pm10

    - platform: mhz19
      co2:
        name: "CO2"
        id: co2
      temperature:
        name: "MH-Z19 Temperature"
      update_interval: 10s
      automatic_baseline_calibration: false
      uart_id: mh
      id: mh_sensor

switch:
  - platform: gpio
    name: "PMS7003_SET"
    pin: 
      number: GPIO26
      mode: OUTPUT
    restore_mode: ALWAYS_ON
    id: pms_set
'''

The readings from the CO2 sensor are just fine and roughly what I expect. However, I get no reading from the PMS7003.

Some observations:

- The fan of the PMS7003 is running, so I assume that 5V and GND are connected correctly
- Using a multimeter, I measured:
  - 0.4 Ohm between the two 5V pins (of PCB header)
  - 0.4 Ohm between the two GND pins (of PCB header)
  - open loop between 5V and GND pins (of PCB header)
  - open loop or >MOhm between other pin combinations (of PCB header)
  - 0.4 Ohm between GPIO1 on Wemos D1 and pin9 of PCB header
From this I conclude that galvanically, the connections should be fine.

I've thought about my PMS7003 being faulty, but I have no way to double check as I have no 1.25 mm pitch headers/converter cables.

I read somewhere online about pulling up the set pin. Is this required for this board?

Is there something else I should check before ordering a new sensor?

DrBlokmeister, 2022-02-27:
I got it working. I wedged a 10 kOhm resistor between 3.3V and GPIO18 and changed the switch configuration to pin18:

```YAML
switch:
  - platform: gpio
    name: "PMS7003_SET"
    pin: 
      number: GPIO18
      mode: OUTPUT
    restore_mode: ALWAYS_ON
    id: pms_set
```

I think your standard YAML file is wrong, as you need to use GPIO18 and not GPIO26.

Otherwise this issue can be closed.

flaviut, 2022-04-06:
Would you mind re-opening this for visibility? I had to do this too, and things now work great.

DrBlokmeister, 2022-04-06:
Reopened as the default yaml file is wrong.

nkitanov, 2024-01-23:
Is that still valid? Sorry, but I do not work actively anymore on this project and can't help much. However, if pins are wrong, then the ESP32 board is different. Not sure if there is any changes with the new ESP32 boards. I remember that RX and TX pins are crossed according to the documentation of the board, even noted that in the documentation:

> In the description of [WeMos_D1_mini_ESP32](https://www.aliexpress.com/item/4000880936043.html?spm=a2g0o.productlist.0.0.f387754egZVnTc&algo_pvid=b105e43b-6669-40bc-99c4-cf22d026995d&algo_expid=b105e43b-6669-40bc-99c4-cf22d026995d-6&btsid=0bb0623616009843570355505e4ca0&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) the pins for UART0 are crossed (TX is where RX is shown). It's easy to fix as you can select those pins but if you carefully check the pins you might wonder why is that.

Badn1nja, 2024-01-24:
Mine is working on GPIO26
[image]

What boards are you using? Maybe it needs another yaml/comment added to yaml just to help anyone in the future with the same board? For reference; I am using MH-ET LIVE esp32 minikit v2.0 with the current yaml

invinciberry, 2025-09-28:
@Badn1nja I tried your fork but no luck. Mind providing a link to where you got your esp32? I suspect it's the board that I bought is different.

invinciberry, 2025-09-29:
@Badn1nja I tried a few more things, and it turns out my issue has something to do with ESPhome updates. Flashing the compiled pin file from your fork fixes my issue entirely, but compiling the firmware from the latest ESPhome version will result in no reading from the PMS sensor.

Do you mind telling me the version you used when compiling the bin file? I can try digging through the change log and try to figure out where the breaking change is.