nkitanov/iaq_board #24: Error in compiling firmware

arjendewolf, 2022-07-08:
I just completed building two of the iaq boards and they both are connected to wifi and work (I love them!). I can integrate them both in Home Assistant independently. However, when I want to integrate them both, it only shows one working device. Since i figured HA has no way to differentiate one from another, I wanted to change the name in the .yaml file. Unfortunately, when compiling I get the error: 

Failed config

light.neopixelbus: [source /config/esphome/arjen.yaml:591]
  
  'variant' is a required option for [light.neopixelbus].
  platform: neopixelbus
  num_leds: 3
  pin: GPIO16
  name: RGB strip
  id: rgb_led
  default_transition_length: 0s

Does someone have an idea how to solve this? Or an alternative approach to solve my initial problem? 
Many thanks!

evilbulgarian, 2022-07-08:
The variant is:
variant: 800KBPS

I had to do this to get them to show up independently in HA using mqtt:

mqtt:
  broker: x.x.x.x
  discovery: true
  discovery_unique_id_generator: mac

Otherwise i think changing the name and using HA api should work just fine.

arjendewolf, 2022-07-08:
> 

Thanks for your quick reply and help! Much appreciated. I am not sure where to put the "Variant: 800KBPS" in the yaml file. 
Any thoughts on that? 
If this should not work, i'll try to dive into mqtt, but that's kind of new to me.

arjendewolf, 2022-07-08:
Thank you! I found out where to put the "variant: 800KBPS" line. Now it starts to compile and it makes lots of progress. Then the following error occurs:

In file included from /data/iaq_boardnieuw/.piolibdeps/iaq_boardnieuw/NeoPixelBus/src/internal/DotStarGenericMethod.h:126:0,
                 from /data/iaq_boardnieuw/.piolibdeps/iaq_boardnieuw/NeoPixelBus/src/NeoPixelBus.h:83,
                 from /data/iaq_boardnieuw/.piolibdeps/iaq_boardnieuw/NeoPixelBus/src/internal/NeoGamma.cpp:28:
/data/iaq_boardnieuw/.piolibdeps/iaq_boardnieuw/NeoPixelBus/src/internal/TwoWireSpiImple.h:29:17: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://registry.platformio.org/search?q=header:SPI.h
*
*************************************************************

compilation terminated.
*** [/data/iaq_boardnieuw/.pioenvs/iaq_boardnieuw/lib7e9/NeoPixelBus/internal/NeoGamma.cpp.o] Error 1
In file included from /data/iaq_boardnieuw/.piolibdeps/iaq_boardnieuw/NeoPixelBus/src/internal/DotStarGenericMethod.h:126:0,
                 from /data/iaq_boardnieuw/.piolibdeps/iaq_boardnieuw/NeoPixelBus/src/NeoPixelBus.h:83,
                 from /data/iaq_boardnieuw/.piolibdeps/iaq_boardnieuw/NeoPixelBus/src/internal/NeoPixelAnimator.cpp:27:
/data/iaq_boardnieuw/.piolibdeps/iaq_boardnieuw/NeoPixelBus/src/internal/TwoWireSpiImple.h:29:17: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://registry.platformio.org/search?q=header:SPI.h
*
*************************************************************

compilation terminated.
*** [/data/iaq_boardnieuw/.pioenvs/iaq_boardnieuw/lib7e9/NeoPixelBus/internal/NeoPixelAnimator.cpp.o] Error 1

A bit strange because I am following the precise steps from previous solved issue in this GIT. 
Any thoughts?

arjendewolf, 2022-07-08:
> Thank you! I found out where to put the "variant: 800KBPS" line. Now it starts to compile and it makes lots of progress. Then the following error occurs:
>
> In file included from [...]

I was able to fix it by removing the entire "platformio options" section. 
Now both boards show up in HA as different devices, but some sensors are not available. When I connect them separately, all sensors work. I'll try some changes in the yaml file and see if that works. Otherwise, your mqtt option is perhaps the most viable one.

arjendewolf, 2022-07-08:
I got it working! Changed the names of the sensors, and now the devices are seperated in HA. So happy now, thanks for your help!

closed 2022-07-08 by arjendewolf