nkitanov/iaq_board #27: Firmware doesnt compile on 2022.12.X ESPHome - linking problem

Dinth, 2022-12-22:
Hi. After the update to 2022.12 the firmware stopped compiling.
```
Compiling /data/iaq-kitchen/.pioenvs/iaq-kitchen/FrameworkArduino/esp32-hal-uart.c.o
Compiling /data/iaq-kitchen/.pioenvs/iaq-kitchen/FrameworkArduino/firmware_msc_fat.c.o
Compiling /data/iaq-kitchen/.pioenvs/iaq-kitchen/FrameworkArduino/libb64/cdecode.c.o
Compiling /data/iaq-kitchen/.pioenvs/iaq-kitchen/FrameworkArduino/libb64/cencode.c.o
Compiling /data/iaq-kitchen/.pioenvs/iaq-kitchen/FrameworkArduino/main.cpp.o
Compiling /data/iaq-kitchen/.pioenvs/iaq-kitchen/FrameworkArduino/stdlib_noniso.c.o
Compiling /data/iaq-kitchen/.pioenvs/iaq-kitchen/FrameworkArduino/wiring_pulse.c.o
Compiling /data/iaq-kitchen/.pioenvs/iaq-kitchen/FrameworkArduino/wiring_shift.c.o
Archiving /data/iaq-kitchen/.pioenvs/iaq-kitchen/libFrameworkArduino.a
Linking /data/iaq-kitchen/.pioenvs/iaq-kitchen/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/lib/libvfs.a(vfs.c.obj): in function `esp_vfs_utime':
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/vfs/vfs.c:585: multiple definition of `utime'; /data/iaq-kitchen/.pioenvs/iaq-kitchen/src/main.cpp.o:/config/esphome/.esphome/build/iaq-kitchen/src/main.cpp:68: first defined here
collect2: error: ld returned 1 exit status
*** [/data/iaq-kitchen/.pioenvs/iaq-kitchen/firmware.elf] Error 1
========================= [FAILED] Took 59.34 seconds ==================
```

fraserp, 2022-12-28:
Experiencing the same, have cleaned the build files to no avail.

paveldn, 2022-12-28:
Change line 427 in iaq_board.yaml:

```
    - platform: uptime
      name: Uptime
      id: utime
```
to 

```
    - platform: uptime
      name: Uptime
      id: utime_sensor

```
worked for me

fraserp, 2023-01-04:
Confirmed working

Dinth, 2023-01-04:
Works for me too. 
Would you mind submitting a PR?

Badn1nja, 2024-01-20:
I just realised this is still an issue and have created this PR

nkitanov, 2024-01-23:
> I just realised this is still an issue and have created this PR

Thanks for your contribution. I can't test your code now, but if you can confirm it is building with the merged yaml I will close this issue.

Badn1nja, 2024-01-23:
It's paveldn's change I have just made the PR - but yes It's working on the latest ESPHome, I have used this change since early 2023

closed 2024-01-23 by nkitanov

Badn1nja, 2024-01-23:
Sorry I should have made another branch for testing.... I have made a PR to fix this version..